ORNLSlicer/ORNLSlicer #19: Feature Request - Replace STL

aroschli, 2023-12-22:
## Expected Behavior
This would open up a dialog box where the user could browse to a new STL and load that. ALL of the STL-specific settings that were applied to the original STL would be carried over to the new STL.

## Actual Behavior
N/A

## Possible Solution
When right-clicking on an STL there is an option to reload part, however, it would be optimal to have an option to "Replace STL."

Many times the user wishes to update an STL to one with a new name, yet retain all of the STL specific settings. Currently, the user has to delete the original STL, load a new STL, and then hope to remember all of the STL specific settings to change. This would eliminate the possible user error of forgetting to modify some settings.

AESjay, 2024-01-11:
@aroschli  @mborish  - this has definitely become a med/high priority for us, as we routinely update STLs with new files and it is tedious and prone to errors on the user's end if they have to manually re-input all the STL specific settings.

aroschli, 2024-05-17:
Functionality added by Cameron

closed 2024-05-17 by aroschli