ORNLSlicer/ORNLSlicer #31: Infill and Skin Exceeding STL Bounds

aroschli, 2024-01-18:
## Expected Behavior
Self-Explanatory.

## Actual Behavior
[image]

## Possible Solution
Self-Explanatory

## Steps to Reproduce the Problem
Self-Explanatory. 

## Specifications
Platform, machine specs, anything else you think might be relevant.
[1530-0205-Test01.zip](https://github.com/ORNLSlicer/Slicer-2/files/13978584/1530-0205-Test01.zip)

AESjay, 2024-01-24:
@aroschli - I thought this was only infill and skins, but I did see this happen last week with insets. Whole project was sliced with about 4 insets, but the tip of the print was all only 2 bead wall (perimeters).  X-section of STL looks correct. But during a print, layers 461 thru 469 all the sudden had insets and starting printing over air.  

[image]

[image]

aroschli, 2024-01-24:
Thanks @AESjay for the additional images. I meant to include inset in the title, but my original image also captures some instances of insets going out of the bounds.

AESjay, 2024-01-24:
OK good, I just hadn't seen it do it with insets until then, so I wanted to
make you aware.

*Jay Evanovich*
800-210-9801 x703
Additive Engineering Solutions, LLC.

aroschli, 2024-02-01:
Issue has been resolved and will be part of the next release.

closed 2024-02-01 by aroschli

AESjay, 2024-02-01:
Excellent, thanks Alex.