ORNLSlicer/ORNLSlicer #42: Cannot compile the project under Linux

klochowicz, 2024-03-20:
## Expected Behavior

It seems that all the dependencies are portable, so the project could run on Linux.

## Actual Behavior

I have hit a compilation error (after installing all the dependencies that CMake informed me about).

```
 fatal error: private/qzipreader_p.h: No such file or directory
   28 | #include <private/qzipreader_p.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
```

## Possible Solution

Maybe some library is missing? I've installed all the CMake requirements... maybe there's an implicit dependency that was not declared with `find_package()` ?

## Steps to Reproduce the Problem
- Try to compile the project under Linux

## Specifications
Ubuntu 22.04

# Labels (delete this section once you've selected labels)

cadkin, 2024-03-20:
Hi, it looks like you are missing some packages on your system. `qzipreader_p.h` is part of the private Qt headers that are not installed by default with the standard Qt installation on Ubuntu.

We have a page on building for Linux [here](https://github.com/ORNLSlicer/Slicer-2/wiki/Building-on-Linux). Check there for the full list of dependencies on Ubuntu. We also have an option for Nix based workflows, but I wouldn't recommend it currently unless you are familiar with that tool.

klochowicz, 2024-03-21:
thank you very much; my bad, it didn't occur to me to look at the right-side of the document!

I have managed to compile the code. thank you for prompt response.

closed 2024-03-21 by klochowicz