ORNLSlicer/ORNLSlicer #61: Spiralize After 1st Layer

aroschli, 2024-09-04:
## Expected Behavior
The ability to print the first layer normally, maybe even with infill/skin, before starting spiralize.

## Actual Behavior
Spiralize runs for the entire print, on every layer.

## Possible Solution
Make spiralize a layer specific setting and/or add a setting for first layer to spiralize, then all future layers are spiralize

## Steps to Reproduce the Problem
Self-Explanatory.

AESjay, 2024-09-27:
I like this idea, and if implemented, should not be limited to spiralizing after only layer 1. It should be user defined (spiralize after layer 1, or 100, etc...)

aroschli, 2024-12-26:
Spiralize added as option for layer specific settings