ORNLSlicer/ORNLSlicer #65: Total adjusted time in the output file

sven3550, 2025-01-08:
The adjusted build time is missing the the beginning comment section of the output file

sven3550, 2025-01-10:
I am using "Marlin Pellet" which only outputs the unadjusted expected build time:
```
;Sliced by: Sven
;Slicing notes: First 6 layers with inset, seam on the back side
;Sliced on: 01/10/2025
;Expected Weight: 1663.81 g
;Expected Build Time: 5 min 39 sec
;Minimum Layer Time: 0 sec
;Maximum Layer Time: 15 sec
```
In this example, the adjusted build taken from the "Layer Timers" tab in the UI was:
```
Total print time: 5 min 40 sec
Total adjusted time: 1 hr 10 min 6 sec
```

aroschli, 2025-01-10:
Which version are you using? This output was adjusted in the 1.0.1 release in August 2024. If you update to the latest version, it should use the adjusted layer time.

aroschli, 2025-01-21:
@sven3550 Just wanted to check in and see if you got this working.

aroschli, 2025-01-22:
Can you provide me a project file to try and recreate the issue?

sven3550, 2025-01-23:
There is a constant layer time enforced via the Material/Cooling settings
[Example_wall_win_1_0_1.zip](https://github.com/user-attachments/files/18516501/Example_wall_win_1_0_1.zip)

aroschli, 2025-01-23:
Thanks, @sven3550 
When I slice and export the g-code, this is the header of the file:
```
;Sliced by: 
;Slicing notes: 
;Sliced on: 01/23/2025
;Expected Weight: 2175.55 g
;Expected Build Time: 1 hr 38 min 26 sec
;Minimum Layer Time: 0 sec
;Maximum Layer Time: 4 min 14 sec
```

If this is what you're expecting, then I believe the issue is resolved. If this is not what you're looking for, let me know and we can drill deeper.

closed 2025-03-05 by aroschli