ORNLSlicer/ORNLSlicer #70: Multiple lines of text in Gcode header description do not get ; added

cmdwyer3, 2025-02-28:
## Expected Behavior
When adding an optional description in the export window to the Gcode header, multiple lines of text should each include a ; symbol to indicate a comment in the Gcode.

## Actual Behavior
Currently, only the first row of text includes a semicolon and the rest are interpreted as normal Gcode.

[image]

[image]

## Possible Solution
Add semicolons to each row of text automatically.

## Steps to Reproduce the Problem

- Slice basic part
- Export part and then add description in the export window popup. Make the description multiple lines of text.
- Check Gcode file after export for the description at the top and notice the lack of ; symbols to denote comments.

## Specifications
Windows 11, Slicer 2 v1.0.1

# Labels (delete this section once you've selected labels)
A complete explanation can be found on the wiki: https://github.com/ORNLSlicer/Slicer-2/wiki/Submitting%20Issues. In short, try to select at least a label identifying where the issue is (usually starts with "core"), a priority, and whether the issue is a bug/feature/wishlist.

cmdwyer3, 2025-02-28:
core-gcode
low-priority
bug

aroschli, 2025-03-04:
Issue has been resolved and will be part of the next release