ORNLSlicer/ORNLSlicer #76: "Settings" tab on right does not size correctly on laptop screen

quintberg, 2025-04-02:
## Expected Behavior
When expanded to Full Screen, the "Settings" tab on the right hand should shrink so it is bounded by the right side of the screen and you don't have to drag the screen to the right. 

## Actual Behavior
When in Full Screen, the "Settings" tab on the right hand of the screen doesn't size itself correctly when viewed on a laptop. It seemingly extends too far to the right so you can't see the values on the far right.  Also, when minimized to a smaller window, you can't shrink the window size using the mouse cursor to try to manually fit the entire window inside your screen. The other tabs show correctly. 

## Possible Solution
Implement the same resolution or sizing code used in the "G-code Editor" and other tabs that show correctly??

## Steps to Reproduce the Problem
1. Open ORNL Slicer 2 on a laptop (single screen)
2. Expand to Full Screen
3. Select the "Settings" tab on the right

## Specifications
Dell Precision 7560 Laptop

# Labels (delete this section once you've selected labels)
A complete explanation can be found on the wiki: https://github.com/ORNLSlicer/Slicer-2/wiki/Submitting%20Issues. In short, try to select at least a label identifying where the issue is (usually starts with "core"), a priority, and whether the issue is a bug/feature/wishlist.

aroschli, 2025-04-04:
Hi, @quintberg . Thank you for submitting an issue. We are aware of this issue. Until we come up with a solution, we recommend changing the display scaling for Windows back to 100%. This issue seems to only happen with high resolution displays set at scaling values of greater than 100%

quintberg, 2025-04-04:
Great, thanks Alex!
-Quinton

This email originated from outside of K-State.

cmdwyer3, 2025-04-18:
I actually found a work-around for this since I ran into a similar issue with the export template window on my laptop. Right click on your slicer 2 shortcut and select "Properties". Then in the "Compatibility" tab click on "Change high DPI settings" at the bottom. Then select override high DPI scaling behavior. Scaling performed by Application. Then start up your slicer and see if that solves the scaling issue.

[image]

aroschli, 2025-04-21:
Thanks for the suggestion @cmdwyer3 I'm going to test this on a few computers and see if this should be offered as an official solution.

quintberg, 2025-04-21:
Yes, thank you @cmdwyer3! It worked on my laptop @aroschli .